Makerfabs/Lora-Soil-Moisture-Sensor #1: Connect to LoraWan / TheThingsNetwork

jdegger, 2020-11-03:
Hi,

I'm experimenting with my sensor, however it is not completely clear to me how I would proceed with connecting the sensor to LoraWan. Is this a possibility of completely out of scope of this project?

Thanks.

Makerfabs, 2020-11-04:
Hi, 

I do not know if this email can find you… seems this is a github message…

 

The sensor is a Lora node, based on 328P and Lora module SX127x, for LoraWAN, theoretically, it is OK, that you program the 328P with firmware that supports LoraWAN, but as I know, it should be not easy.

Basically, we treat this sensor as a separate lora node, without lorawan support. 

 

Best Regards

-----------------------------
Kevin
www.makerfabs.com

jdegger, 2020-11-04:
Thank you Kevin, I am receiving your reply just fine.

Would makerfabs be willing to look into LoraWan support as it's a very interesting technique to add to the support list?

Makerfabs, 2020-11-04:
Er, no.. it would be difficult to support the lorawan.   Actually, if you need Lorawan, there already some IC/module that integrated the lorawan protocol

jdegger, 2020-11-04:
> Actually, if you need Lorawan, there already some IC/module that integrated the lorawan protocol
Where can I find those? Thank you for your quick response

Makerfabs, 2020-11-04:
I believe this: CMWX1ZZABZ -078
https://www.mouser.com/ProductDetail/Murata-Electronics/CMWX1ZZABZ-078?qs=sGAEpiMZZMu3sxpa5v1qrgywWChs744Wl3PkA8jSqjc%3D

jdegger, 2020-11-25:
Hi, I just wanted to let you guys know that I found out that this certainly possible. I'm on my way to make this work but I'm still having some issues. Once I have a working .ino I will share it here. I'm using https://github.com/mcci-catena/arduino-lmic to make this work.

It might be interesting for you as well to use and test this, as it might sell better with LoraWan

lzomedia, 2021-03-29:
> Hi, I just wanted to let you guys know that I found out that this certainly possible. I'm on my way to make this work but I'm still having some issues. Once I have a working .ino I will share it [...]

Interesting I'm keeping my tows in the same library I've managed to get it to 

41630351: EV_TXSTART
41316692: Unknown event: 20
41316966: EV_JOIN_FAILED

mikelinton, 2021-04-16:
@jdegger I'm running into the same issue as I thought I'd be able to use these with LoRaWan/TTN. Were you ever able to get things running with TTN? Thanks in advance.

lzomedia, 2021-04-18:
So I've spent like a whole weekend trying to get it working using a library on ttn. My main problem was the fact that I don't know the pins to listen and send for the lora protocol. Indeed one to one of their thingy works . Even with the pihat again it works. I guess if they can share the schematics we can go forward and maybe could connect this to ttn. But I doubt it. More then sure the pins for in and out are being use.

lzomedia, 2021-04-18:
@mikelinton as you can see from my testing I was having issues with mapping the pins.that library will work perfectly if they can just share the pi outs . I've tried the default ones for the 128 chip and it seems they don't work.

tdvantine, 2021-12-28:
@lzomedia have you looked at the [PDF in the hardware folder](https://github.com/Makerfabs/Lora-Soil-Moisture-Sensor/blob/master/V3/hardware/Lora%20Soil%20Moisture%20Sensor%20v1.2.PDF)? It has a schematic of the circuits.
I too bought these thinking they would work with LoRaWAn, and once I them, i also realized there was no FCCID on them and to me that was a sign they would never be supported for LoRaWAN officially.

mconlow, 2022-01-29:
I also bought one of these thinking I could hook it up to LoRaWAN/TTN...

If anyone gets it working, I'd be very interested. 

Has anyone found another LoRaWAN soil moisture sensor?

mconlow, 2022-01-31:
Another path might be to forego the LoraWAN and just try to use this thing P2P over LoRa which would work in my use-case. 

I have the "bonnet" for the Raspberry Pi https://www.adafruit.com/product/4074
and separately I have a LoRaWAN gateway https://www.adafruit.com/product/4284

Any suggestions from anyone on a python library I could use that would work on either of these devices that could pick up the packet from this moisture sensor?

Thanks in advance