Makerfabs/Lora-Soil-Moisture-Sensor #4: the unit will not program

seamaster101, 2022-01-17:
I can compile your sample, but when i try to upload the upload fails. 
I'm using this programmer: [https://www.amazon.ca/DSD-TECH-Adapter-FT232RL-Compatible/dp/B07BBPX8B8](url)
the programmer is set to 3.3V
I tried connecting the DTR to the reset line on the programmer and to the DTR line on the programmer (it has pad an the board that I pulled out for that)
in both cases the programming fails, but when the unit is powered I can see communication on the serial port:
*********************
Soil start.
success!
AHT10 acknowledged.
success!
ADC:888
ADC:894
ADC:894
BAT:1006
3.24VBAT:1006
3.24VBAT:1006
3.24VID010209 REPLY : SOIL INEDX:0 H:39.58 T:25.84 ADC:894 BAT:1006
[Set]Sleep Mode Set
[Watch dog]0
***********************
etc. etc.....
I have the board setup as Arduino UNO with processor ATmega328P
here is the error message when i try to upload code:
***************************************************************************************************
Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "Arduino Nano, ATmega328P"

Sketch uses 23508 bytes (76%) of program storage space. Maximum is 30720 bytes.

Global variables use 1028 bytes (50%) of dynamic memory, leaving 1020 bytes for local variables. Maximum is 2048 bytes.

avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x53

avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x6f

avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x69

avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x6c

avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x20

avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x73

avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x74

avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x61

avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x72

avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x74

An error occurred while uploading the sketch
**********************************************************************

please tell me what i'm doing wrong?

littlejogi, 2022-01-19:
I select "Arduino Pro or Pro Mini" and ATMega328P (3,3V 8MHZ) and use https://www.amazon.de/gp/product/B01C2P9GD2/ref=ppx_yo_dt_b_asin_title_o03_s01?ie=UTF8&psc=1
Everything works fine. Sometimes i have some problems with the contact of the pin

roopesh, 2023-06-03:
How did you compile this?  I set the board settings (thank you!  They shouldn't make it so hard to figure out) but I still get
````
/Users/*****/Documents/Arduino/Lora-Soil-Moisture-Sensor-V3/Lora-Soil-Moisture-Sensor-V3.ino: In function 'void Lora_init()':
Lora-Soil-Moisture-Sensor-V3:86:82: error: 'SX127X_SYNC_WORD' was not declared in this scope
     int state = radio.begin(FREQUENCY, BANDWIDTH, SPREADING_FACTOR, CODING_RATE, SX127X_SYNC_WORD, OUTPUT_POWER, PREAMBLE_LEN, GAIN);
                                                                                  ^~~~~~~~~~~~~~~~
/Users/*****/Documents/Arduino/Lora-Soil-Moisture-Sensor-V3/Lora-Soil-Moisture-Sensor-V3.ino:86:82: note: suggested alternative: 'RADIOLIB_SX127X_SYNC_WORD'
     int state = radio.begin(FREQUENCY, BANDWIDTH, SPREADING_FACTOR, CODING_RATE, SX127X_SYNC_WORD, OUTPUT_POWER, PREAMBLE_LEN, GAIN);
                                                                                  ^~~~~~~~~~~~~~~~
                                                                                  RADIOLIB_SX127X_SYNC_WORD
Lora-Soil-Moisture-Sensor-V3:87:18: error: 'ERR_NONE' was not declared in this scope
     if (state == ERR_NONE)
                  ^~~~~~~~
/Users/*****/Documents/Arduino/Lora-Soil-Moisture-Sensor-V3/Lora-Soil-Moisture-Sensor-V3.ino:87:18: note: suggested alternative: 'SKIP_NONE'
     if (state == ERR_NONE)
                  ^~~~~~~~
                  SKIP_NONE
exit status 1
'SX127X_SYNC_WORD' was not declared in this scope
````

roopesh, 2023-06-06:
For anyone looking for a solution, I was able to fix it with:
````#include <SPI.h>
#include <Wire.h>
#include <RadioLib.h>
#include <avr/wdt.h>
#include <avr/sleep.h>
#include "I2C_AHT10.h"

#define ERR_NONE 0

//#define NODENAME "LORA_POWER_1"

//IMPORTANT Uncomment out this line with the right ID on the case or on the side of the sensor
//String node_id = String("ID") + "010933";

//Set sleep time, when value is 1 almost sleep 20s,when value is 450, almost 1 hour.
#define SLEEP_CYCLE 450

//Lora set
//Set Lora frequency
//Commented out any frequency in this example to make you choose the relevant one for your sensor.
//IMPORTANT Uncomment out this line with the right Frequency.
//As an aside, I was able to get all of the sensor to broadcast at 915MHz (even the ones marked 434 and 868).
//#define FREQUENCY 434.0
//#define FREQUENCY 868.0
//#define FREQUENCY 915.0

//Nothing below this line should need to be touched.
#define BANDWIDTH 125.0
#define SPREADING_FACTOR 9
#define CODING_RATE 7
#define OUTPUT_POWER 10
#define PREAMBLE_LEN 8
#define GAIN 0

//328p
#define DIO0 2
#define DIO1 6

#define LORA_RST 4
#define LORA_CS 10

#define SPI_MOSI 11
#define SPI_MISO 12
#define SPI_SCK 13

//pin set
#define VOLTAGE_PIN A3
#define PWM_OUT_PIN 9
#define SENSOR_POWER_PIN 5
#define ADC_PIN A2

#define DEBUG_OUT_ENABLE 1

SX1276 radio = new Module(LORA_CS, DIO0, LORA_RST, DIO1);
AHT10 humiditySensor;

bool readSensorStatus = false;
int sensorValue = 0; // variable to store the value coming from the sensor
int batValue = 0;    // the voltage of battery
int count = 0;
int ADC_O_1;           // ADC Output First 8 bits
int ADC_O_2;           // ADC Output Next 2 bits
int16_t packetnum = 0; // packet counter, we increment per xmission
float temperature = 0.0;
float humidity = 0.0;

bool AHT_init()
{
    bool ret = false;
[... 243 lines cut ...]
    readSensorStatus = false;
    digitalWrite(SENSOR_POWER_PIN, LOW); // Sensor/RF95 power off
    delay(100);
}

void all_pins_low()
{
    pinMode(PWM_OUT_PIN, INPUT);
    pinMode(A4, INPUT_PULLUP);
    pinMode(A5, INPUT_PULLUP);

    delay(50);
}

void send_lora()
{
    String message = "INEDX:" + (String)packetnum + " H:" + (String)humidity + " T:" + (String)temperature + " ADC:" + (String)sensorValue + " BAT:" + (String)batValue;
    String back_str = node_id + " REPLY : SOIL " + message;

#if DEBUG_OUT_ENABLE
    Serial.println(back_str);
#endif

    radio.transmit(back_str);
}````

fredmccr, 2024-01-23:
Thank you roopesh for your code.
Unfortunately, I get the following error message while trying to compile it:

"text section exceeds available space in board
Sketch uses 31062 bytes (101%) of program storage space. Maximum is 30720 bytes"

I first tried to remove all the comments lines, hoping it will lighten the weight of the code .. without success.
I finally managed to get rid of this by commenting a few "serial.println..." lines that are relevant for debugging purposes only.
I hope this could help those with the same issue.

roopesh, 2024-01-23:
> I first tried to remove all the comments lines, hoping it will lighten the weight of the code .. without success. I finally managed to get rid of this by commenting a few "serial.println..." lines that are relevant for debugging purposes only. Compilation and programming are ok now. I hope this could help those with the same issue.

hahaha I didn't have that problem, but yeah remove all my extra dumb logging.  I am an Arduino newbie so I needed lots of extra help.