Makerfabs/Lora-Soil-Moisture-Sensor #5: ADC: sensorValue

i4things, 2022-02-09:
Any table that based on the sensor value to get the soil moisture in % ?

littlejogi, 2022-02-11:
i use 
mprocent =(1-((moisture-560) / 324 ))*100;

I don't know how accurate this value is. For me it looks ok

i4things, 2022-02-11:
I use : mpercent = 100 - ((moisture - 500) / 4)
e.g.  500 = 100% i - 900 = 0%

but after some experiments - it looks like a better calculation should use the battery voltage also - as the battery voltage is used as a reference voltage and when the battery goes down the min/max values are changing   

e.g.  battery ADC result with moisture ADC result

bat - 1023 -> moist ( 527(in water -100%) - 894 (in air - 0%) )
bat -  920  -> moist ( 508 - 880 )

e.g. I was hoping someone before me t have done some more detailed experiments :)