Makerfabs/Lora-Soil-Moisture-Sensor #15: Sketch too large

lhvu85, 2024-09-19:
I'm having issues compiling Moisture_LowPower_RadioLib_v2.1.ino.  After correcting SX127X and setting ERR_NONE to 0, it says "Sketch uses 33434 bytes (108%) of program storage space. Maximum is 30720 bytes." (sketch is too big).  I tried removing all the comments like someone else mentioned, but it wasn't enough. 

Also which board are we suppose to use?  Is it Arduino Pro or Pro Mini and choose ATmega328p 3.3V?   Does someone have a working .ino file you can share? Thanks!

ljteam, 2024-10-02:
Same issue here, very frustrating :(

ljteam, 2024-10-02:
Replying to myself if this can help you @lhvu85 : I managed to compile the sketch by selecting an older verson of the Radiolib library (I guess it was smaller before)

cHeeSaW, 2024-11-04:
can anyone of you tell me which version you used? :)

ljteam, 2024-11-04:
Yep, I used Radiolib v6.4.2
[image]